davidker/unisys #119: No kerneldoc for static functions

TurningWrenches, 2017-01-18:
```
 > }
 > 
 > /**
 >  * initialize_controlvm_payload_info() - init controlvm_payload_info struct
 ```
 kerneldoc for a static function?  Ick, no.
 
```
 > }
 > 
 > /**
 >  * visorchipset_chipset_notready() - sends chipset_notready action
 >  *
 >  * Send ACTION=offline for DEVPATH=/sys/devices/platform/visorchipset.
 >  *
 >  * Return: CONTROLVM_RESP_SUCCESS
 ```
 kerndoc for a static function? No.
 
```
 >  */
 > static int
 > visorchipset_chipset_notready(void)
 > {
 > 	kobject_uevent(&visorchipset_platform_device.dev.kobj, KOBJ_OFFLINE);
 ```
 And the kerndoc is toally wrong compared to the code :(
 
 You were better off not documenting it, it would have at least done what
 you said it would do...

``` 
 > }
 > 
 > /**
 >  * read_controlvm_event() - retreives the next message from the
 >  *                          CONTROLVM_QUEUE_EVENT queue in the controlvm
 >  *                          channel
 >  * @msg: pointer to the retrieved message
 >  *
 >  * Return: true if a valid message was retrieved or false otherwise
 >  */
 ```
 kerndoc :(

TurningWrenches, 2017-01-18:
To look for potential violations, I performed the following ```grep``` search on the Unisys driver directory and methodically went through each result.

```
grep -rn "\/\*" drivers/staging/unisys
```

TurningWrenches, 2017-02-03:
**Questions**
```
* visorhba_main.c/visorhba_queue_command_lck()
```

TurningWrenches, 2017-02-03:
These patches are not directly related to the GitHub issue, but are important related changes:
0ee96ac6640f67507037d87240bd98134d88e60d
cf6ce47088e0c35ad5a1d563c95c9d0a7c930e91
04dffe94800c3691697ac473d8310f4a0e16adca

These patches address the GitHub issue:
7ecd703b2bd002f6f4436b502577ba224dd40acd
6e7eda8b43659826cc9a6979c4d4126b2a7694bf
52e15eda80c9077e8a04a5cafa933651ac33a73d
04cf74aaece4c169c7196201f0e3a171a866bee6
f0e5fdf52d355286eadd6eeb531a511c3f13998e
e8f8c803e01a87b6b957a9af0b58703deaa599a3

davidker, 2017-03-10:
Greg accepted this.

closed 2017-03-10 by davidker