davidker/unisys #120: Convert device to misc device instead of platform device (or use acpi device?)

wadgaonkarsam, 2017-01-19:
[Kanboard:24488](http://ustr-erl-6690.na.uis.unisys.com/?controller=task&action=show&task_id=24488&project_id=37)

wadgaonkarsam, 2017-01-19:
This is the initial commit to add the new structure visorchipset_device

Branch: [githubissue-119](https://github.com/davidker/unisys/tree/githubissue-119)
Commit(s): 01568feef9f0021d69feb10302236f817ecfcecf
checkpatch-cleared: Yes
T710-W1 verified: No

wadgaonkarsam, 2017-01-19:
David could you please look at the patch? I am running the T710-W1 test now. Should I run any additional tests for this patch.

selltc, 2017-01-19:
I added a few minor comments directly on commit 01568feef9f0021d69feb10302236f817ecfcecf.

I also assume there are more patches to come for this issue.

wadgaonkarsam, 2017-01-20:
Thanks Tim and David. I updated the patch and the branch(I was on #119 instead of #120). The T710-W1 passed this time.

Branch: [githubissue-120](https://github.com/davidker/unisys/tree/githubissue-120)
Commit(s): 3f0454ce18974f640da7e5edab603937fa44e472
checkpatch-cleared: Yes
T710-W1 verified: Yes

wadgaonkarsam, 2017-01-26:
Branch: githubissue-120
Commit(s): 4d015df 8f41b8b 8e35ae0
checkpatch-cleared: Yes
T710-W1 verified: Yes

wadgaonkarsam, 2017-02-01:
Branch: [githubissue-120](https://github.com/davidker/unisys/tree/githubissue-120)
Commit(s):  f0d0a9dbcebf1d0f685585ac5853ce2b19132c28 80137afb4cecbaa8a2ef417159da6fe7b2656713
checkpatch-cleared: Yes
T710-W1 verified: Yes

davidker, 2017-02-01:
I made a comment directly on 80137afb4cecbaa8a2ef417159da6fe7b2656713

Looks good, I just think one line needs to be changed.

davidker, 2017-02-08:
Sameer, for the first patch, I don't see where you are freeing it or if you free on it. We might need to add that to the patch.

davidker, 2017-02-08:
I attempted to fix it by this patch: 5e306c56d1a0cf9b754ea5c73088ecf4c5e92056

wadgaonkarsam, 2017-02-08:
Thank you David, I'll add that to the first patch.

wadgaonkarsam, 2017-02-14:
Branch: [githubissue-120](https://github.com/davidker/unisys/tree/githubissue-120)
Commit(s):  
48b452ae9803b3923e93eba7953bcb1dc2985807
2bc8d037ea4a37ae41a9d86c7e4e9daa9d556f69
f0f833fac639bde685da2db7d52a5590b4118154
checkpatch-cleared: Yes
T710-W1 verified: Yes

davidker, 2017-02-15:
Sameer, patch f0f833f leaves the following unused #define VISORCHIPSET_MMAP_CONTROLCHANOFFSET can you remove that as well as part of this patch?

wadgaonkarsam, 2017-02-15:
Branch: [githubissue-120](https://github.com/davidker/unisys/tree/githubissue-120)
Commit(s):  
4b945420d7357948ef7f429f7076f298ca2bc69d
2e87a533de74e088576eae1231ac73ef9361f914
4bb509fa1f43b978564e7dcdede721f3e9c658b5
checkpatch-cleared: Yes
T710-W1 verified: Yes

Edited the second and the third patch. Added the visorchipset_dev_groups back to use sysfs_create_groups and removed the unused #define and #includes.

wadgaonkarsam, 2017-02-16:
Branch: [githubissue-120](https://github.com/davidker/unisys/tree/githubissue-120)
Commit(s):  
a9155e92629c854a3dce0ceca9278d84816f6a88
d669a4400945a31054e0d77de1145a90861eb2de
3a26c974f2edcced1c69fc3fd1d9307013ca72c1
checkpatch-cleared: Yes
T710-W1 verified: Yes

wadgaonkarsam, 2017-02-18:
Branch: [githubissue-120-test](https://github.com/davidker/unisys/tree/githubissue-120-test)
Commit(s): 33b57212dfcc3a019696e6c1bae9f0e56a9a702c 
checkpatch-cleared: Yes
T710-W1 verified: Yes

davidker, 2017-03-10:
Greg accepted this.

closed 2017-03-10 by davidker