davidker/unisys #124: Nits in initialize/destroy_controlvm_payload_info()

davidker, 2017-01-27:
Task #24543

davidker, 2017-01-27:
Realized they aren't actually being used just created and destroyed. So we can get rid of it. Patch is located on branch [Githubissue-124-upstream-next)[https://github.com/davidker/unisys/tree/githubissue-124-upstream-next] with the patch: 08a82575d406cb74e17ccfcd1a61923e9f2693ee.

TurningWrenches, 2017-01-27:
So I understand what your patch is doing, and it looks fine to me. My only comment is to refine the commit message a bit more. When I read it, I don't make the connection that visor_controlvm_payload_info has anything to do with sending payloads back to the s-Par firmware. I might say something to the effect of "Removing visor_controlvm_payload_info structure, which was used to send payloads back to the s-Par firmware, because we do not implement that functionality."

davidker, 2017-02-06:
Committed at d308d2b

closed 2017-02-06 by davidker