davidker/unisys #125: Clean up #defines (right hand)

TurningWrenches, 2017-02-03:
```
Right hand alignment needs to be cleaned up and consistent. 

 > #define POLLJIFFIES_CONTROLVMCHANNEL_FAST   1
 > #define POLLJIFFIES_CONTROLVMCHANNEL_SLOW 100
 
 Right-hand alignment?  That's a glutton for punishment.
```

TurningWrenches, 2017-02-03:
```
grep -rn "#define" drivers/staging/unisys/
drivers/staging/unisys/visorbus/visorchannel.c:29:#define MYDRVNAME "visorchannel"
drivers/staging/unisys/visorbus/visorchannel.c:31:#define SPAR_CONSOLEVIDEO_CHANNEL_PROTOCOL_GUID \
drivers/staging/unisys/visorbus/visorchannel.c:166:#define SIG_QUEUE_OFFSET(chan_hdr, q) \
drivers/staging/unisys/visorbus/visorchannel.c:174:#define SIG_DATA_OFFSET(chan_hdr, q, sig_hdr, slot) \
drivers/staging/unisys/visorbus/visorchannel.c:182:#define SIG_WRITE_FIELD(channel, queue, sig_hdr, FIELD)			 \
drivers/staging/unisys/visorbus/visorbus_main.c:24:#define MYDRVNAME "visorbus"
drivers/staging/unisys/visorbus/visorbus_main.c:31:#define LINESIZE 99
drivers/staging/unisys/visorbus/visorbus_main.c:33:#define CURRENT_FILE_PC VISOR_BUS_PC_visorbus_main_c
drivers/staging/unisys/visorbus/visorbus_main.c:34:#define POLLJIFFIES_NORMALCHANNEL     10
drivers/staging/unisys/visorbus/controlvmchannel.h:16:#define __CONTROLVMCHANNEL_H__
drivers/staging/unisys/visorbus/controlvmchannel.h:22:#define SPAR_CONTROLVM_CHANNEL_PROTOCOL_UUID	\
drivers/staging/unisys/visorbus/controlvmchannel.h:26:#define ULTRA_CONTROLVM_CHANNEL_PROTOCOL_SIGNATURE \
drivers/staging/unisys/visorbus/controlvmchannel.h:28:#define CONTROLVM_MESSAGE_MAX 64
drivers/staging/unisys/visorbus/controlvmchannel.h:36:#define ULTRA_CONTROLVM_CHANNEL_PROTOCOL_VERSIONID  1
drivers/staging/unisys/visorbus/controlvmchannel.h:38:#define SPAR_CONTROLVM_CHANNEL_OK_CLIENT(ch)           \
drivers/staging/unisys/visorbus/controlvmchannel.h:47:#define CONTROLVM_QUEUE_REQUEST		0
drivers/staging/unisys/visorbus/controlvmchannel.h:48:#define CONTROLVM_QUEUE_RESPONSE	1
drivers/staging/unisys/visorbus/controlvmchannel.h:49:#define CONTROLVM_QUEUE_EVENT		2
drivers/staging/unisys/visorbus/controlvmchannel.h:50:#define CONTROLVM_QUEUE_ACK		3
drivers/staging/unisys/visorbus/controlvmchannel.h:53:#define CONTROLVM_CRASHMSG_MAX		2
drivers/staging/unisys/visorbus/controlvmchannel.h:464:#define CONTROLVM_RESP_SUCCESS                                  0
drivers/staging/unisys/visorbus/controlvmchannel.h:465:#define CONTROLVM_RESP_ALREADY_DONE                             1
drivers/staging/unisys/visorbus/controlvmchannel.h:466:#define CONTROLVM_RESP_IOREMAP_FAILED                           2
drivers/staging/unisys/visorbus/controlvmchannel.h:467:#define CONTROLVM_RESP_KMALLOC_FAILED                           3
drivers/staging/unisys/visorbus/controlvmchannel.h:468:#define CONTROLVM_RESP_ID_UNKNOWN                               4
drivers/staging/unisys/visorbus/controlvmchannel.h:469:#define CONTROLVM_RESP_ID_INVALID_FOR_CLIENT                    5
drivers/staging/unisys/visorbus/controlvmchannel.h:472:#define CONTROLVM_RESP_CLIENT_SWITCHCOUNT_NONZERO               100
drivers/staging/unisys/visorbus/controlvmchannel.h:473:#define CONTROLVM_RESP_EXPECTED_CHIPSET_INIT                    101
drivers/staging/unisys/visorbus/controlvmchannel.h:476:#define CONTROLVM_RESP_ERROR_MAX_BUSES		201	/* BUS_CREATE */
drivers/staging/unisys/visorbus/controlvmchannel.h:477:#define CONTROLVM_RESP_ERROR_MAX_DEVICES        202	/* DEVICE_CREATE */
drivers/staging/unisys/visorbus/controlvmchannel.h:479:#define CONTROLVM_RESP_PAYLOAD_INVALID		400	/* SWITCH_ATTACHEXTPORT,
drivers/staging/unisys/visorbus/controlvmchannel.h:482:#define CONTROLVM_RESP_INITIATOR_PARAMETER_INVALID 401  /* Multiple */
drivers/staging/unisys/visorbus/controlvmchannel.h:483:#define CONTROLVM_RESP_TARGET_PARAMETER_INVALID    402  /* DEVICE_CONFIGURE */
drivers/staging/unisys/visorbus/controlvmchannel.h:484:#define CONTROLVM_RESP_CLIENT_PARAMETER_INVALID    403  /* DEVICE_CONFIGURE */
drivers/staging/unisys/visorbus/controlvmchannel.h:486:#define CONTROLVM_RESP_BUS_INVALID                 500	/* SWITCH_ATTACHINTPORT,
drivers/staging/unisys/visorbus/controlvmchannel.h:492:#define CONTROLVM_RESP_DEVICE_INVALID           501 /* SWITCH_ATTACHINTPORT */
drivers/staging/unisys/visorbus/controlvmchannel.h:497:#define CONTROLVM_RESP_CHANNEL_INVALID          502 /* DEVICE_CREATE,
drivers/staging/unisys/visorbus/controlvmchannel.h:501:#define CONTROLVM_RESP_VIRTPCI_DRIVER_FAILURE   604       /* BUS_CREATE,
drivers/staging/unisys/visorbus/controlvmchannel.h:507:#define CONTROLVM_RESP_VIRTPCI_DRIVER_CALLBACK_ERROR 605  /* BUS_CREATE,
drivers/staging/unisys/visorbus/controlvmchannel.h:513:#define CONTROLVM_RESP_GENERIC_DRIVER_CALLBACK_ERROR 606
drivers/staging/unisys/visorbus/controlvmchannel.h:521:#define CONTROLVM_RESP_ERROR_BUS_DEVICE_ATTACHED 700	/* BUS_DESTROY */
drivers/staging/unisys/visorbus/controlvmchannel.h:523:#define CONTROLVM_RESP_CHANNEL_TYPE_UNKNOWN 800	        /* GET_CHANNELINFO,
drivers/staging/unisys/visorbus/controlvmchannel.h:526:#define CONTROLVM_RESP_CHANNEL_SIZE_TOO_SMALL 801	/* DEVICE_CREATE */
drivers/staging/unisys/visorbus/controlvmchannel.h:528:#define CONTROLVM_RESP_CHIPSET_SHUTDOWN_FAILED            1000
drivers/staging/unisys/visorbus/controlvmchannel.h:529:#define CONTROLVM_RESP_CHIPSET_SHUTDOWN_ALREADY_ACTIVE    1001
drivers/staging/unisys/visorbus/controlvmchannel.h:532:#define CONTROLVM_RESP_CHIPSET_STOP_FAILED_BUS            1100
drivers/staging/unisys/visorbus/controlvmchannel.h:533:#define CONTROLVM_RESP_CHIPSET_STOP_FAILED_SWITCH         1101
drivers/staging/unisys/visorbus/controlvmchannel.h:536:#define CONTROLVM_RESP_DEVICE_UDEV_TIMEOUT                1400
drivers/staging/unisys/visorbus/visorchipset.c:32:#define CURRENT_FILE_PC VISOR_BUS_PC_visorchipset_c
[... cut ...]
drivers/staging/unisys/include/channel.h:402:#define SPAR_VHBA_CHANNEL_PROTOCOL_UUID_STR \
drivers/staging/unisys/include/channel.h:406:#define SPAR_VNIC_CHANNEL_PROTOCOL_UUID \
drivers/staging/unisys/include/channel.h:411:#define SPAR_VNIC_CHANNEL_PROTOCOL_UUID_STR \
drivers/staging/unisys/include/channel.h:415:#define SPAR_SIOVM_UUID \
drivers/staging/unisys/include/channel.h:421:#define SPAR_CONTROLDIRECTOR_CHANNEL_PROTOCOL_UUID  \
drivers/staging/unisys/include/channel.h:429:#define SPAR_DIAG_POOL_CHANNEL_PROTOCOL_UUID				\
```

TurningWrenches, 2017-02-10:
**Branch:** githubissue-125-upstream-next-TurningWrenches
**Commit(s):** 
5c78b84cea1ba091210171b764f331c09d6e74bc
d290704a367018b448b1f4fa6e379d1103233940
97206f845853d50c0bc9655aa77881d17d0f6f18
aa24fbb3e1a90c2920ff463f9b24888a64d71f19
b49a0a4caff08a064d3a7141aeb8296877e1a7cc
f08f5490cac0e740800462781477dc7ee1dc088e
bb3df060ea42c1b8ff666a4c59f31c1690cbe227
51d081d37582122abd3919bb91e0283271d32435
fc3b361bccd5f980b875acc3d46259d47fe50126
fc7325a5ad8da161c4e33064e5740cf81bdeeb95
9e5db22c2bed40a11156045c70350de267b07338
**checkpatch-cleared:** No
**T710-1 verified (smoke-test):** No

TurningWrenches, 2017-06-26:
Is this now extinct? @davidker

davidker, 2017-06-26:
We have a patch in for-later that gets rid of them, though Greg wants us to keep some of them due to them being hardware/protocol defines (even though we don't use them currently)

davidker, 2017-07-27:
I'm closing this since all changes are in upstream-next or no longer needed.

closed 2017-07-27 by davidker