davidker/unisys #128: Change function names to be consistent with driver namespace

wadgaonkarsam, 2017-04-21:
[Kanboard:29383](http://ustr-erl-6690.na.uis.unisys.com/?controller=task&action=show&task_id=29383&project_id=37)
``` 
void
bus_create_response(struct visor_device *bus_info, int response)
{
 	if (response >= 0)
 		bus_info->state.created = 1;
```
 And that's a horrid global function name, please use your own namespace.
```
}
 
void
bus_destroy_response(struct visor_device *bus_info, int response)
```
same here, use your own namespace.
```
{
 	bus_responder(CONTROLVM_BUS_DESTROY, bus_info->pending_msg_hdr,
 		      response);
 
 	kfree(bus_info->pending_msg_hdr);
 	bus_info->pending_msg_hdr = NULL;
}
 
void
device_create_response(struct visor_device *dev_info, int response)
```
same namespace issue.
```
}

void
device_destroy_response(struct visor_device *dev_info, int response)
```
namespace...
```
{
	device_responder(CONTROLVM_DEVICE_DESTROY, dev_info->pending_msg_hdr,
			 response);

	kfree(dev_info->pending_msg_hdr);
	dev_info->pending_msg_hdr = NULL;
}

void
device_pause_response(struct visor_device *dev_info,
		      int response)
```
namespace...
```
{
	device_changestate_responder(CONTROLVM_DEVICE_CHANGESTATE,
				     dev_info, response,
				     segment_state_standby);

	kfree(dev_info->pending_msg_hdr);
	dev_info->pending_msg_hdr = NULL;
}

void
device_resume_response(struct visor_device *dev_info, int response)
```
namespace, errors, etc...

wadgaonkarsam, 2017-04-25:
Branch: [githubissue-128](https://github.com/davidker/unisys/tree/githubissue-128)
Commit(s): 
1775f2106706dfbbba2a3fb63c8ebf3e4870a3c6
ed9a2d7e264748dd7886a383da14a6d6fff53996
4cbf9173f535092caf1cea20fe7b4288a6bf84b9
9c059e713a2c54d3b4f0fe9e7bde6c843272e9e7
0a4899e3a88854b963626420d6e0782f8c19807a
7eb1b3a4a96cba49b16d119a7cf7af3a1b4cd876
7606b2cece740a6e5a73be0ba54e4266840ffa5b
checkpatch-cleared: Yes
T710-W1-AutoTest verified: Yes

davidker, 2017-05-02:
Sameer, I'm going through channel.h and I see a lot of places where describing with namespace can improve it. I'm wondering if we should look at this file as well.

davidker, 2017-05-04:
Patch: staging: unisys: visorbus: vbuschannel.h: renamed #defines and structures to match driver namespace

Can you split it up into #defines and structures?

davidker, 2017-05-04:
Same with the controlvmchannel.h one as well.

davidker, 2017-05-04:
And for channel.h (even though it makes one patch extremely small)

davidker, 2017-05-10:
I noticed a couple of things with the current githubissue-128 branch. We missed the Documentation directory and I think we should drop protocol from the channel names.