davidker/unisys #135: Simplify visorchipset functions

wadgaonkarsam, 2017-07-17:
[KanBoard:30818](http://ustr-erl-6690.na.uis.unisys.com/?controller=task&action=show&task_id=30818&project_id=37)
Merge "response" function into create/destroy function, and then call the create/destroy function directly instead of the visorchipset function.

wadgaonkarsam, 2017-07-18:
Branch: [githubissue-135](https://github.com/davidker/unisys/tree/githubissue-135)
Commits:
36ff610593f679e0ca7319467137fc9a80a605ef
08bb4c0efcc5f0a5cfeb3c9f88318e3fb1a133cf
56ed7d0aa97c86f04dd2b8f35a06d94feaeb0c59
Checkpatch: Passed
T710-W1-AutoTest: Passed

davidker, 2017-07-18:
I didn't get this to apply cleanly to upstream-next. Might be the changes that I just added :(