davidker/unisys #138: Clean up visor_check_channel

wadgaonkarsam, 2017-08-07:
[KanBoard:30856](http://ustr-erl-6690.na.uis.unisys.com/?controller=task&action=show&task_id=30856&project_id=37)

visor_check_channel should solely belong to visorbus and not published to other drivers.

The structure channeltype_descriptor should be expanded to accept the parameters visor_check_channel needs (except for SIGNATURE which is the same for everyone). And during driver registration the structure should be filled out and during device discovery it should be tested.

Also we should add a channeltype_descriptor for controlvm channel cause it is a visorchannel too, right?

wadgaonkarsam, 2017-08-08:
Branch: [githubissue-138](https://github.com/davidker/unisys/tree/githubissue-138)
Commits:
d5a3ce36cf9b70bab926ae8df9748be437fba45d
ea2d7ad4ef0ec8e32c86e70d4cfb45e381e7d618
90845e3934c983b0463d1a56cac03c30a1a1dcf2
69e6c0788bad409b3cf5121b1aa05e60f6d2f62e
Checkpatch: Passed
T710W1-AutoTest: Passed

davidker, 2017-08-22:
Doesn't apply to upstream-testing, can you rebase and try again.

wadgaonkarsam, 2017-08-22:
Branch: [githubissue-138](https://github.com/davidker/unisys/tree/githubissue-138)
Commits:
2cdd827fdade318014339f2b7250e14114834885
10d9e7cfa1fe754aa6cf9c3dc71372d895cca1a7
a9be7771111d0608eda893b0f21d13f79bed07a3
cc5bb92e4c124c93fe20f048602c9fbfdea3d4f2
Checkpatch: Passed
T710W1-AutoTest: Passed

wadgaonkarsam, 2017-08-23:
Thanks David! Updated patch 3 and fixed a typo in the patch comment in patch 4.

Branch: [githubissue-138](https://github.com/davidker/unisys/tree/githubissue-138)
Commits:
2cdd827fdade318014339f2b7250e14114834885
10d9e7cfa1fe754aa6cf9c3dc71372d895cca1a7
507fc99c232a297c1fccdff471d11c30c5500b90
d1ffebd8b7b72af2be89487922a39448d415510c
Checkpatch: Passed
T710W1-AutoTest: Passed